aleppax/AirQualityMesh #22: pico not exiting lightsleep but hanging

aleppax, 2023-03-10:
The problem is known https://github.com/micropython/micropython/issues/9006
But the solution suggested by felixdoerre doesn't seem to solve all the problems.
Someone should try to debug with swd

aleppax, 2023-03-12:
https://github.com/tomjorquera/pico-micropython-lowpower-workaround/blob/master/lowpower.py
Looking into this could be helpful

aleppax, 2023-03-12:
While doing more tests, I realized that the low level of current during lightsleep can't occur when the LED pin output is initialized. Therefore, since we would have, but currently do not use it anyway, I'll remove it from the init process.

aleppax, 2023-03-14:
this is blocking. The problem still occurs with "&& cyw43_poll != NULL" in modmachine.c
after many minutes (?) that REPL connection is lost and is unresponsive, thonny console resets and shows REPL prompt, but pico doesn't execute main.py as usual. I'm not using boot.py

aleppax, 2023-03-16:
18 hours without lightsleep stopping, it seems to me that at least in this installation the problem is solved, I await confirmation from @robferrero to close this issue.

robferrero, 2023-03-16:
I'm still testing , actually I find useful info here too  https://forum.micropython.org/viewtopic.php?f=21&t=12657&p=68842&hilit=sleep+wlan.deinit+wlan.deinit#p68842

aleppax, 2023-05-12:
another few days and if the problem doesn't occur any more I'll close this issue.

closed 2023-05-17 by aleppax